OpenRoberta/robertalab #1012: Revise Lists in all robots: Java

boonto, 2018-09-28:
Parent issue: #1011 
- [ ] create empty list
- [ ] create list with content
- [ ] create list with repeated element
- [ ] length of
- [ ] is empty
- [ ] find `first`
- [ ] find `last`
- [ ] get `#`
- [ ] get `# from end`
- [ ] get `first`
- [ ] get `last`
- [ ] get and remove `#` 
- [ ] get and remove `# from end`
- [ ] get and remove `first`
- [ ] get and remove `last`
- [ ] remove `#`
- [ ] remove `# from end`
- [ ] remove `first`
- [ ] remove `last`
- [ ] insert at `#`
- [ ] insert at `# from end`
- [ ] insert at `first`
- [ ] insert at `last`
- [ ] get sub-list from `#` to `#`
- [ ] get sub-list from `#` to `# from end`
- [ ] get sub-list from `#` to `last`
- [ ] get sub-list from `# from end` to `#`
- [ ] get sub-list from `# from end` to `# from end`
- [ ] get sub-list from `# from end` to `last`
- [ ] get sub-list from `first` to `#`
- [ ] get sub-list from `first` to `# from end`
- [ ] get sub-list from `first` to `last`
- [ ] sum
- [ ] min
- [ ] max
- [ ] average
- [ ] median
- [ ] standard deviation
- [ ] for each
- [ ] assignment

boonto, 2018-09-28:
Artem &#128152; Patrick: Here are the suggested one-liner (except standard deviation) implementations:

```Java
List<Float> emptyList = new ArrayList<>();
System.out.println(emptyList);

List<Float> specifiedList = new ArrayList<>(Arrays.asList(1.0f, 2.0f, 3.0f));
System.out.println(specifiedList);

List<Float> repeatedList = new ArrayList<>(Collections.nCopies(5, 1.0f));
System.out.println(repeatedList);

int lengthOf = repeatedList.size();
System.out.println(lengthOf);

boolean isEmpty0 = repeatedList.isEmpty();
System.out.println(isEmpty0);

int findFirst = IntStream.range(0, repeatedList.size()).filter(i -> repeatedList.get(i) == 0.0f).findFirst().orElse(-1);
System.out.println(findFirst);

int findLast = IntStream
    .range(0, specifiedList.size())
    .filter(i -> specifiedList.get(i) == 3.0f)
    .reduce((first, second) -> second)
    .orElse(-1);
System.out.println(findLast);

int index = 2;
float getNum = specifiedList.get(index);
float getNumFromEnd = specifiedList.get((specifiedList.size() - 1) - index);
float getFirst = specifiedList.get(0);
float getLast = specifiedList.get(specifiedList.size() - 1);

specifiedList.addAll(Collections.nCopies(20, 0.0f));
float getAndRemoveNum = specifiedList.remove(index);
float getAndRemoveNumFromEnd = specifiedList.remove((specifiedList.size() - 1) - index);
float getAndRemoveFirst = specifiedList.remove(0);
float getAndRemoveLast = specifiedList.remove(specifiedList.size() - 1);

specifiedList.remove(index); //removeNum
specifiedList.remove((specifiedList.size() - 1) - index); //removeNumFromEnd
specifiedList.remove(0); //removeNumFirst
specifiedList.remove(specifiedList.size() - 1); //removeNumLast

float num = 2.0f;
specifiedList.add(index, num); //insertAtIndex
specifiedList.add((specifiedList.size() - 1) - index, num); //insertFromEnd
specifiedList.add(0, num); //insertFirst
specifiedList.add(num); //insertLast

int ind0 = 2, ind1 = 3;
specifiedList.subList(ind0, ind1);
specifiedList.subList(ind0, (specifiedList.size() - 1) - ind1);
specifiedList.subList(ind0, (specifiedList.size() - 1));
ind1 = 18;
specifiedList.subList((specifiedList.size() - 1) - ind0, ind1);
ind1 = 1;
specifiedList.subList((specifiedList.size() - 1) - ind0, (specifiedList.size() - 1) - ind1);
specifiedList.subList((specifiedList.size() - 1) - ind0, (specifiedList.size() - 1));
specifiedList.subList(0, ind1);
specifiedList.subList(0, (specifiedList.size() - 1) - ind1);
specifiedList.subList(0, (specifiedList.size() - 1));

List<Float> mathList = new ArrayList<>(Arrays.asList(0.0f, 1.0f, 2.0f, 0.0f, -1.0f, 0.1f, 12.0f));
float sum = (float) mathList.stream().mapToDouble(Float::floatValue).sum();
float min = mathList.stream().min(Float::compareTo).get();
float max = mathList.stream().max(Float::compareTo).get();
float average = (float) mathList.stream()
    .mapToDouble(Float::floatValue)
    .sum() / mathList.size();
float median = (mathList.size() % 2) == 0 ?
    (float) mathList.stream()
        .sorted()
        .skip(mathList.size() / 2 - 1)
        .limit(2)
        .mapToDouble(Float::floatValue)
        .average()
        .getAsDouble() :
    mathList.stream()
        .sorted()
        .skip(mathList.size() / 2)
        .findFirst()
        .get();
System.out.println(median);
float standardDeviation = 0.0f; //skip

List<Float> loopList = new ArrayList<>(Arrays.asList(0.0f, 1.0f, 2.0f));
for (Float f : loopList) {
    System.out.println(f);
}
System.out.println(loopList);

List<Float> assList0 = new ArrayList<>(Arrays.asList(0.0f, 1.0f, 2.0f));
List<Float> assList1 = new ArrayList<>(Arrays.asList(3.0f, 4.0f, 5.0f));
assList0 = assList1;
System.out.println(assList0);
System.out.println(assList1);
assList0.add(0.0f);
System.out.println(assList0);
System.out.println(assList1);
```

Needed discussion:
- Do we need standard deviation?
- **all one liners** vs **all abstracted** vs **complicated ones abstracted** (e.g. findLast, median)?

boonto, 2018-10-02:
During discussion we decided to add all of the list functions as one liners.
The complicated math functions should be abstracted into visible and easily readable functions in the program.

boonto, 2018-10-09:
The EV3 does not support Java 8 without additional work from the user (and it seems unstable), so the Java 8 specific implementations will be replaced by (if still possible) one liners or readable functions.

The numeric list type cannot be easily created. E.g.
```java
List<Float> list = new ArrayList<>(Arrays.asList(1, ((float) 1.1), 2, 3))
```
does not work as `1`, `2` and `3` are integers. This may also cause more problems in the future. Possible solutions:
- multi line array initialisation (may only be a temporary solution)
- split numeric into int/float (this affects all projects)
- cast all numbers in Java to float (may cause other problems)
- remember the current code state, to conditionally generate different numbers in the number evaluation (complicates a lot of things)

boonto, 2018-11-16:
Most of the functionality is now implemented using standard Java methods. The remaining usages of BlocklyMethods will be replaced in #1016. 
The random operation can be removed for all robots at once in #1011.

VinArt, 2018-11-26:
Hi, @RohitMidha23, this issue is tagged 'resolved' because the implementation is already provided and is currently on the develop branch, but needs checking, so that there are no bugs. If you wish, you can do some testing to make sure that the lists implementation for java-based robots is working correctly and if you encounter problems or bugs, let us know in this issue, or if you want to discuss something, then here https://gitter.im/open-roberta-lab/Testing

rbudde, 2018-11-29:
I generated code for the sum, average, median, std deviation, random. Looks fine.

closed 2019-03-23 by bjost2s