OpenRoberta/robertalab #1015: Revise Lists in all robots: C++

VinArt, 2018-10-02:
Parent issue: #1011 

- [ ] create empty list
- [ ] create list with content
- [ ] create list with repeated element
- [ ] length of
- [ ] is empty
- [ ] find `first`
- [ ] find `last`
- [ ] get `#`
- [ ] get `# from end`
- [ ] get `first`
- [ ] get `last`
- [ ] get and remove `#` 
- [ ] get and remove `# from end`
- [ ] get and remove `first`
- [ ] get and remove `last`
- [ ] remove `#`
- [ ] remove `# from end`
- [ ] remove `first`
- [ ] remove `last`
- [ ] insert at `#`
- [ ] insert at `# from end`
- [ ] insert at `first`
- [ ] insert at `last` # treat as append in code generation, naming has to be revised
- [ ] get sub-list from `#` to `#` # treat as from `#` to `#` exclusive
- [ ] get sub-list from `#` to `# from end` 
- [ ] get sub-list from `#` to `last`  # check current implementation
- [ ] get sub-list from `# from end` to `#` # check current implementation
- [ ] get sub-list from `# from end` to `# from end` # check current implementation
- [ ] get sub-list from `# from end` to `last` # check current implementation
- [ ] get sub-list from `first` to `#` # check current implementation
- [ ] get sub-list from `first` to `# from end` # check current implementation
- [ ] get sub-list from `first` to `last` # check current implementation
- [ ] sum
- [ ] min
- [ ] max
- [ ] average
- [ ] median
- [ ] standard deviation # if used generate a standard deviation function
- [ ] for each
- [ ] assignment
- [ ] access random elements in lists # should not be supported anymore

bjost2s, 2018-10-02:
What about calliope?

VinArt, 2018-10-19:
All the C++ based robots can support the same implementation, calliope uses stl and arduino based robots ArduinoSTL, they both have the same interface, so the implementation is needed only once

VinArt, 2018-10-19:
C++ related functionality is implemented and put to the NEPODefs.h for RobotArdu resources as well as for the RobotMbed resources. Functions were tested with standalone C++ programs. Currently GUI blocks are revised, needed ones added and unwanted options (random) removed.

VinArt, 2018-11-05:
Calliope lists implemented, need to check arduino further. All commits form the feature branch were squashed into two: the one with the most of implementation and the other containig blockly copy and semmicolon fix.

bjost2s, 2018-12-04:
Please check this again, the following program is not compilable:
[image]
/home/bjost/git/robertalab/OpenRobertaParent/RobotArdu/resources/libraries/RobertaFunctions/NEPODefs.h:95:29: error: invalid cast from type ‘String’ to type ‘const char*’
         if ((P) (*iterator) == value) {
                             ^

bjost2s, 2018-12-05:
still another problem!
[image]

VinArt, 2018-12-05:
this:
[image]
will, however, fail. And I am not sure I can do anything about it right now. Although I cannot find a use case when one would want to insert a value in a temporary list and immediately throw the whole thing away.

closed 2019-03-23 by bjost2s