OpenRoberta/robertalab #1017: Arduino write analog/digital program destroyeed in the front-end

rbudde, 2018-10-02:
**Describe the bug**
 - A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a configuration with LED on D0
2. Write a program, that sets digital pin D13 on, then wait, then off
3. save this program. D13 is reset to D0, digital is reset to analog
4. The same happens if you view the generated source

**Expected behavior**
 - The user selected pin name should persist

**Screenshots**
 [image]
- my test program

**Device information**
 - arduino nano

**Additional context**
 - looks like a refresh problem with blockly

**Definition of Done**
- the user selected pin name should persist export, see source, run program
- when an exported program is reloaded, it should show the original pin names

sebastiankmilo, 2018-10-06:
and there is another problem, the arduino can´t write in analog pin(pwm), it only can in specific digital pin.
[image]

bjost2s, 2018-10-10:
@sebastiankmilo thank you very much for pointing to this additional problem!

rbudde, 2018-10-10:
the same problem (blockly program vanishes, i.e. is reset to the initial program) for BOB3 programs (see file /OpenRobertaServer/src/test/resources/crossCompilerTests/bob3/eye.xml for instance)

VinArt, 2018-12-04:
The last commit was meant for issue #1071, not #1017

VinArt, 2018-12-05:
Program for bob3 was vanishing for different reason, colours were the problem. Implemented in #1071

commit a7107290 (2018-10-18): fix #1017 create configuration blocks for read/write analog/digital pin, TODO change codegeneration completely
closed 2019-02-11 by VinArt