OpenRoberta/robertalab #1026: Cannot Comment on start block

lotu, 2018-10-12:
While working with middle school students I wanted to use comments the students to use a comment to remind themselves what a program does.  They meet once per week and it is very easy to forget in that time. 

The best place for such a comment is in the start block because it cannot be deleted  during changing the program, and because it could be added before the student write and code.   The latter is a clarifying task to help students solidify their thoughts.   

For the above reasons it would be nice if you could comment on the start block.

lotu, 2018-10-12:
Oh... I just figured out how to use the comment block under Text..........................

bjost2s, 2018-10-15:
There is also a comment possibility on the right panel, click on the 
[image]
icon and use the tiny editor to document the program

closed 2019-02-21 by bjost2s