OpenRoberta/robertalab #1027: the same problem (blockly program vanishes, i.e. is reset to the initial program) for BOB3 programs (see file /OpenRobertaServer/src/test/resources/crossCompilerTests/bob3/eye.xml for instance)

bjost2s, 2018-10-15:
_Originally posted by @rbudde in https://github.com/OpenRoberta/robertalab/issues/1017#issuecomment-428588538_

bjost2s, 2018-10-15:
this issue only appears on the test server. There must have been changes on server side:
STATISTICS {"time":"2018-10-15 21:29:18,427", "sessionId":"1", "robotName":"bob3", "logLevel":"ERROR", "message":"Transformer failed"}
de.fhg.iais.roberta.util.dbc.DbcException: Invalid de.fhg.iais.roberta.mode.general.PickColor: #228822
	at de.fhg.iais.roberta.transformer.Jaxb2BlocklyProgramTransformer.invokeMethod(Jaxb2BlocklyProgramTransformer.java:78)
	at de.fhg.iais.roberta.transformer.Jaxb2BlocklyProgramTransformer.invokeJaxbToAstTransform(Jaxb2BlocklyProgramTransformer.java:67)
	at de.fhg.iais.roberta.transformer.Jaxb2BlocklyProgramTransformer.blockToAST(Jaxb2BlocklyProgramTransformer.java:57)
	at de.fhg.iais.roberta.transformer.Jaxb2BlocklyProgramTransformer.instanceToAST(Jaxb2BlocklyProgramTransformer.java:50)
	at de.fhg.iais.roberta.transformer.Jaxb2BlocklyProgramTransformer.transform(Jaxb2BlocklyProgramTransformer.java:40)
	at de.fhg.iais.roberta.util.jaxb.JaxbHelper.generateProgramTransformer(JaxbHelper.java:115)
	at de.fhg.iais.roberta.transformer.BlocklyProgramAndConfigTransformer.transform(BlocklyProgramAndConfigTransformer.java:77)
	at de.fhg.iais.roberta.javaServer.restServices.all.ClientProgram.command(ClientProgram.java:149)

rbudde, 2018-11-29:
Colors on BOB3 won't work. The design of colours in frontend -> colours in xml2ast -> colors in code gen is very strange. It's not easy to fix this without redesigning the colour processing at all.

The front end puts RGB-like values into the XML. These are sometimes RGB values, sometimes enumerations. Jaxb translates this to strings. The mapping RGB <-> String must be unique for ALL robots for both values of the pair (RGB,String). Code generation takes these constants and uses them for adressing .h constants in BOB3, for example. If either names of the constants or RGB-values like #000000 are in conflict with values of other robots, the transformations fail.

Colours must be refactored.

VinArt, 2018-12-05:
I implemented colours for arduino-based robots in a different way, so these kind of programs do not fail now.