OpenRoberta/robertalab #1029: Decouple the server properties from the robot properties

rbudde, 2018-10-17:
**Is your feature request related to a problem? Please describe.**
 - adding a robot should nor need to update server parts (OpenRobertaServer) or general robot parts (OpenRobertaRobot) 

**Describe the solution you'd like**
 - change the name of all robot-specific property files to <plugin-name>.properties (lower/uppe case problem)
- move all robot-specific properties to the robot-specific property files
- refactor the ServerStarter and the robot-specific factories to work with these changes

**Additional context**
 - improves the structure a lot and decouples components in a nice way (it is works :-)

rbudde, 2018-10-17:
remark: should we integrate the ports.properties into the regular properties? Sounds good.

rbudde, 2018-10-17:
in this context we MUST change the name of the botnroll plugin from the (very misleading) name "ardu" to "botnroll". This is good anyhow.

rbudde, 2018-11-14:
all plugin specific properties are moved to the properties of the plugin. Thus, only the white list of plugins to be activated are kept in the global properties. This solves the issue.

VinArt, 2018-12-20:
updateResources.dir property is still in server properties.

rbudde, 2019-02-12:
working. Will create a ticket for an update server of its own and put that into backlog.

commit 98888531 (2018-11-28): fix #1029 and #1065 from blockly
closed 2019-02-12 by rbudde