OpenRoberta/robertalab #1030: USBProgram: Arduino type is not updated when connecting a different Arduino

boonto, 2018-10-18:
**Describe the bug**
 - After disconnecting an Arduino and connecting a different type, the old type is reused.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the USBProgram and connect any Arduino
2. Connect to the Server and register with the selected type in the lab
3. Disconnect from the server with the USBProgram
4. Disconnect the Arduino from the PC and connect another type (E.g. Mega then Uno)
5. Connect to the server without changing the Arduino type in the lab
6. Observe that the connection still works

**Expected behavior**
 - The connection should not work, the Arduino type should be updated.

rbudde, 2019-02-12:
The first one (i.e. lowest COM number) is taken. But changing DOES work. I'll file an issue assembling enhancements for the USBprogram.

Works.

closed 2019-02-12 by rbudde

boonto, 2019-02-12:
@rbudde there's a [wiki entry](https://github.com/OpenRoberta/robertalab/wiki/Open-Roberta-Robot-Connection-Program) with all kinds of possible USBProgram enhancements. You could add it there.