OpenRoberta/robertalab #1036: Compile error when a variable called 'time' is added to start block of Calliope mini

CMDSpectre, 2018-11-03:
**Describe the bug**
Compile error, when declaring/initializing a global variable called 'time' in the start block of the 'Calliope mini' and 'Calliope mini 2016' in https://lab.open-roberta.org/

**To Reproduce**
Steps to reproduce the behavior:
1. Open https://lab.open-roberta.org/
2. Click on 'Calliope mini' or 'Calliope mini 2016'
3. Add a variable(type seems irrelevant error occurs anyway) called 'time' to the start block
4. Click on 
     > run on Calliope mini

5. The following error message appears 
     > Attention
Please check the error messages in your progamming blocks and the configuration! Your program has errors so it cannot be compiled.

**Expected behavior**
It should compile.

**Screenshots**
[image]

**Device information**
 - Type: Desktop
 - OS: Windows Version 1803 Build 17134.345 (64-bit)
 - Motherboard: Gigabyte Aorus Gaming K5 (BIOS version F23)
 - Processor: AMD Ryzen 2700x (just naming motherboard and processor because I had problems in the past with those and sometimes still do, although this seems unrelated)
 - Browser: Firefox 63.0.1 (64-bit)

**Additional context**
 - First time I am opening an issue ever ;-)
 - It could be a redeclaration in the same the same scope OR attempt to reassign a constant on Calliope mini
 - If it happens with the name time, I would assume there are other possible variable names that also cause conflict
 - No errors are being shown inside the programming blocks

VinArt, 2018-11-05:
Thanks for writing and congratulations on your first issue!

Indeed time is declared in arm-none-eabi/include/time.h:
`time_t     time (time_t *_timer);`
So the compiler throws an error:
```
error: 'double time' redeclared as different kind of symbol
 double time;
arm-none-eabi/include/time.h:59:11: note: previous declaration 'time_t time(time_t*)'
 time_t    time (time_t *_timer);
error: assignment of function 'time_t time(time_t*)'
     time = 0;
```
Function time itself is described [here](http://www.cplusplus.com/reference/ctime/time/), in short - "Get the current calendar time as a value of type time_t."

One of our future goals is to deliver our users better feedback, probably with some parts from the compiler messages. Please keep in mind that Calliope's c++ code is compiled against the standard library, so there can be other possibilities of such collisions.

bjost2s, 2018-11-06:
I will add it to the list of reserved words so that one cannot use "time" as a variable name anymore. Although we cannot add (or better identify) all already used names for variables and functions, "time" is worth to be added there because it is used in this way often.

closed 2018-12-20 by VinArt