OpenRoberta/robertalab #1040: EV3 is totally broken and unusable

lotu, 2018-11-06:
**Describe the bug**
 - When you try to run an EV3 program that robot sits there with a blank screen and does not run the program.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect an EV3 robot
2. Write a simple program
3. Push Run
4. Robot screen turns blank indicating that the program is running but no movement happens.
5. verify robot works but writing equivalent program in LEGO Mindstorms.

If I ssh into the robot and run the program via the command line

I get the following
$ ./NEPOprog.py 
Traceback (most recent call last):
  File "./NEPOprog.py", line 15, in <module>
    'A':Hal.makeLargeMotor(ev3dev.OUTPUT_A, 'on', 'foreward', 'right'),
  File "/usr/lib/python3/dist-packages/roberta/ev3.py", line 48, in makeLargeMotor
    m.polarity = 'normal'
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 486, in polarity
    self._polarity = self.set_attr_string(self._polarity, 'polarity', value)
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 229, in set_attr_string
    return self._set_attribute(attribute, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 216, in _set_attribute
    raise Exception("%s is not connected" % self)
Exception: LargeMotor(outA) is not connected

If I change the first motor form A to B the error changes as well.

**Expected behavior**
 - Robot just runs

**Device information**
robot@ev3dev:~$ apt-cache show openrobertalab
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version
Package: openrobertalab
Version: 1.7.1+1.0.0
Architecture: all
Maintainer: Stefan Sauer <ensonic@google.com>
Installed-Size: 209
Depends: init-system-helpers (>= 1.18~), python3, python3:any (>= 3.3.2-2~), systemd, python3-bluez, python3-dbus, python3-ev3dev, python3-gi
Enhances: brickman
Homepage: http://lab.open-roberta.org/
Priority: optional
Section: python
Filename: pool/main/o/openrobertalab/openrobertalab_1.7.1+1.0.0_all.deb
Size: 28014
SHA256: edf67f516a1c8fb9bccbef0449d0987e8dda2cbbab694a90c53360fb15351814
SHA1: c042479710f9e619d3ab58357f0effc55614b089
MD5sum: d2ae498a07a1884a44b22f5365829ef8
Description: lab.open-roberta.org connector for ev3dev.org
 Open-roberta.org is a web platform for learning how to program robots. This
 package is a service for the LEGO MINDSTORMS EV3 running ev3dev.org to connect
 to the cloud IDE.
Description-md5: bf389c4a084a3e69f47a43af8257788b

**Additional context**
 - I really need a work around immediately I am teaching a class of middle schoolers with this and discovered the bug in class when no-one's robot's worked.  We will be back next week and I need a plan to make this work even if it is ugly.

bjost2s, 2018-11-06:
Have you already tested it with a new intallation of the ev3dev image?

lotu, 2018-11-06:
Yes these where flashed with the ev3dev stretch image release in August.

bjost2s, 2018-11-06:
Thank you.
I currently have no ev3dev image by hand, so maybe @ensonic, can you help?

ensonic, 2018-11-06:
Which version of ev3dev is this. You should still use jessie.

lotu, 2018-11-07:
It is ev3dev stretch, I was flashing new devices, it say says on https://www.ev3dev.org/downloads/ " ev3dev-stretch is now in BETA. This version is recommended for new users!"  So I used that I will try reflashing with jessie and see what happens.

lotu, 2018-11-08:
I see this now.  I tested with ev3 dev jessie and it works fine.

Thank you for the help

closed 2018-11-09 by bjost2s