OpenRoberta/robertalab #1042: [NXT] Certain strings involving quotation marks lead to failed compilation to machine code

Stex, 2018-11-07:
**Describe the bug**

Using a string containing `"Sensor` leads to "The program could not be transformed into machine code."

This was noticed when the text `Please press "Sensor 1"`

**To Reproduce**

1. Create a simple program to display the text `"Sensor`
  [image]

2. Try to send the program to a connected NXT

The generated NXC code looks fine to me, but I haven't be able to get a local compiler running to see if it throws an error.

[image]

Until now, this only happened for `"Sensor`, but I could imagine that there might be other clashes instances. I'd suspect a clash with internal identifiers during the compilation, but that's just a guess.

**Expected behavior**

The program can be compiled and transferred to the NXT.

VinArt, 2018-11-28:
I think, this is an issue with NXC compiler, because I edited the program to have a different string and compiled it to intermediate NBC code, where I later changed it back to \"Sensor and compiled that to the binary, that worked. So as the output states, the preprocessor somehow fails, but I am not sure, whether we can avoid that somehow, except for allowing only single quotation marks in the strings in the frontend.

bjost2s, 2018-11-29:
The text block is generic, so why shouldn't we allow double  quotation marks for all other robots. I would suggest to simply exchange them while generating code, from double to single.

rbudde, 2018-12-10:
This issue definitiely shows a bug/problem in the NXC preprocessor/compiler: the string \"Sensor\" hits a macro definition and doesn't work. Using the strings \"qay\" or even \"sensor\" works fine.

We need a user-friendly way to show in such situations, that cross-compilers fail (e.g. showing their messages). It must be obvious for the user, that not the lab, but the underlying tool failed.