OpenRoberta/robertalab #1047: [NXT] Light sensor lamp is turned on too late (reflected light)

Stex, 2018-11-08:
**Describe the bug**

When using a light sensor block to measure reflected light, the lamp is turned on too late, resulting in a wrong result.

In the program below, a very low value is displayed as the light is measured before the lamp is even turned on. As the lamp takes its time, even the "wait" block is triggered before the reflected light can actually be measured.

This can be fixed by adding any block that uses the light sensor and a `wait` block at the start of the actual program - just to make sure the lamp is turned on in time.

In the original NXT software, the lamp is always turned on at the start of each program that uses it, maybe that would be a good solution here as well (automatically insert a corresponding call at the top of each NXC `main` task if the lamp is used at least once)

**To Reproduce**
1. Write a program that relies on the light sensor for reflected light, e.g. the following:
  [image]

2. Make sure the light sensor is calibrated correctly (it is, unfortunately, I cannot do that through orlab though)
3. Place the robot on a white area in front of a black line, e.g.
  [image]
4. Run the program

**Expected behavior**

1. The robot displays a high number as it's on a white surface
2. The robot starts moving forward
3. The robot stops at the black line

bjost2s, 2018-11-10:
Could you please provide correct source code for this issue? So how should the nxc code looks like to achieve your expectations.

Stex, 2018-11-11:
The Lego NXT-G software by Lego seems to do the following: If the light sensor is used in a program, it waits with the program execution until the light is actually turned on.

I was able to reproduce this behaviour by calling `SetSensorLight(S3, true);` explicitly before the first actual command is executed. Not sure if this is the optimal solution though.

```c
#define WHEELDIAMETER 5.6
#define TRACKWIDTH 12.0
#define MAXLINES 8
#include "NEPODefs.h" // contains NEPO declarations for the NXC NXT API resources

task main() {
    SetSensor(S1, SENSOR_TOUCH);
    SetSensor(S2, SENSOR_SOUND);
    SetSensor(S3, SENSOR_LIGHT);
    SetSensor(S4, SENSOR_LOWSPEED);
    SetSensorLight(S3, true);
    OnFwdReg(OUT_BC, SpeedTest(30), OUT_REGMODE_SYNC);
    while (true) {
        if ( SensorLight(S3, "LIGHT") < 40 ) {
            break;
        }
        Wait(15);
    }
    Off(OUT_BC);
    while (true) {
        if ( Sensor(S1) == true ) {
            break;
        }
        Wait(15);
    }
}
```

VinArt, 2019-02-11:
ResetSensor is called to wait for sensor re-configuration. The timing problem should not occur anymore.

closed 2019-02-11 by VinArt