OpenRoberta/robertalab #1048: [NXT] Cannot initialize empty list in NXC

Stex, 2018-11-08:
**Describe the bug**

Originally, this ticket was about expanding a list in NXC, but then I noticed, this starts even earlier:

Orlab allows me to initialize a variable with an empty list.
This doesn't work in the generated NXC code as its [`ArrayBuild`](http://bricxcc.sourceforge.net/nbc/nxcdoc/nxcapi/group___array_functions_gae8d75e08c670f09f01200a897d7ed2ba.html#gae8d75e08c670f09f01200a897d7ed2ba) function expects the array to fill and **at least** one element.

I'd therefore expect either not being able to create an empty list in orlab or to convert it to a construct in NXC that emulates the desired behaviour.

**To Reproduce**
Steps to reproduce the behavior:

1. Write a program like the following:
  [image]

2. Run it in the simulator
3. Be happy that it works
4. Try to send it to the brick
5. Get a compiler error

The generated NXC code:

```c
#define WHEELDIAMETER 5.6
#define TRACKWIDTH 12.0
#define MAXLINES 8
#include "NEPODefs.h" // contains NEPO declarations for the NXC NXT API resources 

float numbers[0];
task main() {
    float __numbers[] = {};
    numbers = __numbers;
    SetSensor(S1, SENSOR_TOUCH);
    SetSensor(S2, SENSOR_SOUND);
    SetSensor(S3, SENSOR_LIGHT);
    SetSensor(S4, SENSOR_LOWSPEED);
    TextOut(0, (MAXLINES - 1) * MAXLINES, "Hallo");
}

```

The compiler error:

```
# Error: Invalid number of arguments: at least 2 expected, 1 found
File "/home/pi/workspace/bug_expand_empty_list.nxc" ; line 8
#   	arrbuild __main_7qG2___numbers_7qG2_000
```

**Expected behavior**
The program compiles.

VinArt, 2018-11-16:
- Initialisation of arrays with zero elements should (and will) be removed from blockly for nxt.

bjost2s, 2019-02-21:
What is the current state of this issue?