OpenRoberta/robertalab #1059: Revise Lists in all robots: NXC

VinArt, 2018-11-16:
parent issue #1011 
Stub for an issue concerning NXT/NXC

VinArt, 2018-11-22:
I think that Blockly.Msg.LISTS_*_TITTLE for nxc should be replaced with _TITLE_NXC and the titles themselves should not contain the word "list", instead the word "array" should be used.

bjost2s, 2018-11-29:
We cannot easily rename blocks. What if a user has already programs with these blocks.

bjost2s, 2018-11-29:
Please implement the checker so that an already labeled block (see #1060) will stop the compiler workflow and returns the exact error message.

VinArt, 2018-12-03:
To sum up: there were two possibilities to implement the checker - either append the error tag and add the transformation, or add an attribute and change the XML schema. The second is merged to develop, the first is saved on feature/blockly-errors branch.

VinArt, 2019-02-14:
set element from end needs additional -1 to the index.

VinArt, 2019-02-14:
Average Operation on list needed an additional length parameter since the nxc compiler has a bug/feature that does not allow using ArrayLen in assignments and expressions.

closed 2019-02-14 by VinArt

rbudde, 2019-02-15:
code generated for arrays still wrong (failing usages of ArrayLen)

bjost2s, 2019-02-19:
What is the current state of this issue?
I will remove it from project 2.1.0 "to do" because it is already released