OpenRoberta/robertalab #1066: Seeed LED bar for Calliope won't turn off

Dittse, 2018-11-28:
**Describe the bug**
 - Once the LEDs of the LED bar are turned on, you can't turn them off anymore by setting their brightness to 0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://lab.open-roberta.org/#'
2. Click on 'Calliope'
3. Choose the "set LED Bar x brightness" Block to set the Brightness of the LEDs to a number except from 0
4. Set them to 0
5. See the Error

**Expected behavior**
 - LEDs would turn off

**Device information**
 - Type: Laptop
 - OS: Windows 7 Enterprise
 - Browser: Firefox 63.0.3 (64-Bit)

**Additional context**
 - Code seems fine

Dittse, 2018-11-28:
I tried with another Calliope and it worked fine

VinArt, 2018-11-29:
So, does this work or is this still an issue?

Dittse, 2018-11-30:
It depends on the Calliope that is used. I tested it on two (both 2017) and it works on one but not the other

bjost2s, 2019-02-21:
@Dittse , if you find a second not working calliope, please reopen this issue ;-)

closed 2019-02-21 by bjost2s