OpenRoberta/robertalab #1071: Redesign of colour processing from frontend to code generation

rbudde, 2018-11-29:
The design of colours in frontend -> colours in xml2ast -> colors in code gen is very strange. It's not easy to fix this without redesigning the colour processing at all.

**1. colour defs between robots overlap. But all defs are in OpenRobertaRobot**
Colours belong to a robot. There is litte sense for having them defined globally. Colours should be moved to the robot, where they make sense.

**2. ACTUAL design decisions w.r.t. colours are problematic**
The front end puts RGB-like values as #445566 into the XML. These are sometimes RGB values, sometimes enumerations, always written like #FFFFFF or #000001. Jaxb translates this to enum-like strings. Obviously the mapping RGB <-> String must be unique for both values of the pair (RGB,String). These constants are used in code generation (for adressing .h constants in BOB3, for example).

**Proposal**
1. the frontend should put colour strings like "white" or "blue" or RGB#rrggbb into the XML
2. jaxb checks whether these colour strings are legal for the robot, but doesn't change them.
3. Code generation takes the colour strings as they are (for adressing either .h constants or to generate valid RGB values. 

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

VinArt, 2018-12-04:
2f0390e fixes the conversion division error

bjost2s, 2018-12-04:
new problem in mBot:
[image]
/tmp/B3MSQRQH/NEPOprog (1)/source/NEPOprog (1).ino:26:5: error: ‘_rgbled’ was not declared in this scope
     _rgbled.setColor(2, _rgbled.setColor(2, 0xB3, 0x00, 0x06);
     ^

void loop()
{
    _rgbled.setColor(2, _rgbled.setColor(2, 0xB3, 0x00, 0x06);
    _rgbled.show();
}
rgbled.setColor should not be there and there is a bracket missing at the end

VinArt, 2018-12-05:
[image]

VinArt, 2018-12-05:
NXT, Ev3 and WeDo are left to be checked

bjost2s, 2018-12-20:
@VinArt , this issue is in backlog, but you have committed on this.

Is there still something to do?

rbudde, 2018-12-21:
some commits mentioned in this issue are necessary followups of the refactoring, e.g. commit https://github.com/OpenRoberta/robertalab/commit/522329352b82326b5e0482173d99ce892fadf715 . The last commit mentioned by @bjost2s adds tests and is 14 days old. I appreciate the increase of test coverage. Sometimes enhancements hit issues actually out of scope, but it is good, that they are documented.

bjost2s, 2018-12-22:
If this issue is not in the current project it will not be reviewed. I think with the changes above it should be reviewed and then moved to backlog

closed 2019-06-26 by rbudde