OpenRoberta/robertalab #1075: Dotted Notes

Dittse, 2018-12-03:
**Is your feature request related to a problem? Please describe.**
 - If I want to compose musical masterpieces with the Calliope mini, I usually also need dotted notes. 
A dotted note is held 50% longer than its non-dotted version. 
e.g. if a whole note is held for 2000ms, a dotted whole note is held for 3000ms.
However, this is not possible with the simple "Games Note" block and the workaround is too complicated for children.

**Describe the solution you'd like**
 - Add this type of notes in the first drop-down list of the block.

**Describe alternatives you've considered**
 - A checkbox "Dotted [Yes/No]" would also be a conceivable solution.

bjost2s, 2018-12-09:
I am not really sure, if kids do know about dotted notes. If not, I think it is not a problem for advanced users to created dotted notes "by hand" with the "frequence" block.

Dittse, 2018-12-10:
It's a recurring request. Because basically musicality is universally present for everyone and so working with the Calliope is more accessible for everyone.

To display the dotted note duration with the frequency block is in fact not a big challenge. However, one is then also forced to pass the corresponding frequency to the block. 
This is relatively simple for a1 (440 Hz), but a simple c1 has already 261.626 Hz.
(https://de.wikipedia.org/wiki/Frequenzen_der_gleichstufigen_Stimmung)

The point is that the current functionality is very limited and you are already limited in your choice of simple example melodies as a teacher. This small extension would increase the possibilities for beginners many times over without demotivating them.