OpenRoberta/robertalab #1078: Arduino configuration allows selecting wrong pins for components

VinArt, 2018-12-05:
**Describe the bug**
When configuring a component for arduino it is possible to:
- select the same pin number for different outputs (or inputs) of the component like this:
[image]
- select the same pin number for different components like this:
[image]
- select unsuitable pins (for example for analog output it is possible to select pins that do not support PWM) see image above. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to configuration tab of an arduino robot
2. Select unsuitable pins

**Expected behavior**
 - Either a message should be shown that the pin is in use, or available pin set should be reduced
 - Unusable pins should not appear in the list at all

**Additional context**

#756 is related

VinArt, 2018-12-05:
From [Arduino reference](https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/)

> On most Arduino boards (those with the ATmega168 or ATmega328P), this function works on pins 3, 5, 6, 9, 10, and 11. On the Arduino Mega, it works on pins 2 - 13 and 44 - 46. Older Arduino boards with an ATmega8 only support analogWrite() on pins 9, 10, and 11.

bjost2s, 2018-12-09:
In general we haven't implemented a checker for the use of pins so far. The idea was to do it on server side, because there we do it also the "old" configuration.

rbudde, 2019-06-26:
a job for a configuration checker