OpenRoberta/robertalab #1080: Arduino read/write analog/digital pin implementation missing

VinArt, 2018-12-05:
#1017 related

pablopoo, 2018-12-19:
Hi @VinArt, with 2 or more digital or analog actuators, the switch between them don't work.
[image]

pablopoo, 2018-12-19:
Also found that the code generated in the program loop for the actuator variable name is in uppercase.
[image]

VinArt, 2018-12-19:
@pablopoo thank you for the comments. All that you mentioned is already fixed, but not yet released, stay tuned as it will go online soon.

VinArt, 2018-12-19:
@pablopoo if you like, you can also join our gitter channel where we discuss testing: https://gitter.im/open-roberta-lab/Testing it is brand new, but you can write your findings there.

closed 2019-02-11 by VinArt