OpenRoberta/robertalab #1081: AdBlock prevents lab from being loaded

VinArt, 2018-12-05:
**Describe the bug**
 - When AdBlock is used in the browser in prevents the lab page from being loaded because of the cookie disclaimer

**To Reproduce**
Steps to reproduce the behavior:
1. Install adblock for chrome
2. Go to localhost:1999 
3. Logo keeps spinning and page does not load

**Expected behavior**
 - The page would load

**Screenshots**
[image]

**Device information**
 - Desktop
 - OS: Linux
 - Browser: Chrome Version 70.0.3538.77 (Official Build) (64-bit)
 - AdBlock 3.35.0

rbudde, 2018-12-07:
I installed adblock (most people use adblock plus) and activated it. I was unable to reproduce the error:

- loading the lab succeeded
- removing all cookies and loading the lab succeeded

Tested on Windows8 and Chrome Version 70.0.3538.110 (Offizieller Build) (64-Bit).

issue labelled as resolved

bjost2s, 2018-12-09:
Same here with Ubuntu 18.04
Chrome Version 71.0.3578.80 (Official Build) (64-bit)
Cannot be reproduced

closed 2018-12-09 by bjost2s