OpenRoberta/robertalab #1083: Calliope: local variables rename themself

Dittse, 2018-12-07:
**Describe the bug**
 - Functions with local variable change the displayed local variable in use when your definition is placed under another function with local variable. This applies to functions with and without return value in any combination.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Calliope
2. Build the following Programms
3. Compile to Calliope or show code 
4. See the local variable change itself

**Expected behavior**
 - variables should keep their names

**Screenshots**
 - Generating this is fine:
[image]
Generating this will produce the bug:
[image]

**Device information**
    Type: Laptop
    OS: Windows 7 Enterprise
    Browser: Firefox 63.0.3 (64-Bit)

**Additional context**
 - Code seems fine and programs work as expected on real calliope
- This was tested on the Testserver https://test.open-roberta.org//#

bjost2s, 2018-12-08:
Same for other robots and on public server

bjost2s, 2018-12-10:
This problem is hard to solve and needs deep changes in blockly.