OpenRoberta/robertalab #1084: Wedo: Can't select note on ipad

Dittse, 2018-12-10:
**Describe the bug**
 - you can't select another not than c1 on Ipad

**To Reproduce**
Steps to reproduce the behavior:
1. Open the open roberta mobile app
2. get the play note block
3. try to select another note
4. you can't.

**Expected behavior**
 - You can select another note

**Screenshots**
 - If applicable, add screenshots to help explain your problem.
[image]

**Device information**
 - Thorstens IPad 4
 - Open Roberta Mobile App Version 1.0.0

bjost2s, 2018-12-10:
strange, working fine for me on safari Iphone.
please provide some more information of the device you have used

Dittse, 2018-12-17:
It says: 
Version 10.3.3 (14G60)
Modell: MD510FD/A

It is currently in B3-116, if you want to try hands on.

bjost2s, 2018-12-20:
@rbudde , could you please test it? Could be a webview issue and if so, it  should be fixed before the app will be published