OpenRoberta/robertalab #1092: 2 Arduinos cannot be connected to the lab at the same time

rbudde, 2018-12-13:
**Describe the bug**
 - when the usb-program is used to connect 2 different arduinos to the lab, the arduino connected later terminates the connection of the one connected earlier.
- so it is impossible to connect 2 arduinos at the same time.
- the bug occurs in the public lab, but also on dedicated Raspberry Pi's.

**Reason**
- when robots loose connection to the lab, in former times it was necessary to reconnect them by supplying a fresh token (this was often the case with ev3dev)
- to avoid this stupid work, the mac addr of the robot was used as a unique identifier and if a "new connection" was requested, the old was used, if the mac addr matched
- this made the assumption, that mac addr are different world-wide and that they are supplied with each connect call from the robot 
- this was a bit naive and at two places corrections are made:
  - if the mac addr from the robot  (this includes the usb program!) is "usb", the re-use was not done
  - if the mac addr from the robot (this includes the usb program!) was missing at all, the mac addr "1234" was used. In this case "1234" was interpreted later as a world-wide unique mac addr and the re-use took place and the behavior as described above occurs. The usb program omits the mac addr (or sends "1234" :-) if arduinos are connected.

**Correction**
- [x] if the mac addr is missing at all, "usb" is used as the default. The error is remedied.
- [ ] later the usb program should be fixed

Thanks to Artem and Philip to reconstruct the problem on a local machine and, of course, to colleagues from h-brs, which detected the bug

rbudde, 2018-12-13:
It has to be decided on 14.12.2018 whether we hotfix this or not.

bjost2s, 2018-12-13:
+1 for a hotfix

VinArt, 2018-12-13:
Just to search for this quicker, when it is going to be fixed, here's the patch:
```
--- OpenRobertaParent/OpenRobertaServer/src/main/java/de/fhg/iais/roberta/javaServer/restServices/robot/RobotCommand.java
+++ OpenRobertaParent/OpenRobertaServer/src/main/java/de/fhg/iais/roberta/javaServer/restServices/robot/RobotCommand.java
@@ -58,7 +58,7 @@ public class RobotCommand {
         }
         // TODO: move robot to the requested properties for the next version
         String robot = requestEntity.optString("robot", "ev3");
-        String macaddr = requestEntity.optString("macaddr", "1234");
+        String macaddr = requestEntity.optString("macaddr", "usb");
         String brickname = requestEntity.optString("brickname", robot);
         String batteryvoltage = requestEntity.optString("battery", "");
         String menuversion = requestEntity.optString("menuversion", "");
```