OpenRoberta/robertalab #1093: Nepo4Arduino: Startup popup dont work if only one arduino plugin is active

pablopoo, 2018-12-13:
**Describe the bug**
If only one arduino plugin is active (-d robot.whitelist=sim,uno -d robot.default=uno), the the startup plugin show the message "Robot not found" under the nepo4arduino icon.

**To Reproduce**
1.-Start the server with sim and uno plugin only: -d robot.whitelist=sim,uno -d robot.default=uno
2.-go to the startup popup

**Expected behavior**
 The popup screen must show the active arduino plugin. It works right now with the 3 plugins active (uno, nano, mega) or any combination of 2, but not if only one is active.

**Screenshots**

-1 active -d robot.whitelist=sim,uno -d robot.default=uno:

[image]

[image]

-2 active -d robot.whitelist=sim,uno,mega -d robot.default=uno:

[image]

[image]

pablopoo, 2018-12-21:
@bjost2s I'm thinking in classrooms where only one kind of robot is available, i.e. with arduino uno only.
If was implemented that way, a comment here [https://github.com/OpenRoberta/robertalab/blob/d321ddd1c4def2a5c3b973c3fc30dda1d326e86a/OpenRobertaParent/OpenRobertaServer/src/main/resources/openRoberta.properties#L54](url) is enough for other developers.

rbudde, 2019-06-26:
Let us think about a new start popup, which handles the special case of a single robot. The popup is overloaded anyhow with the many robots we support now. This should be not too much effort