OpenRoberta/robertalab #1102: It is not possible to load a program link export for micro:bit

philippmaurer, 2018-12-21:
**Describe the bug**
When creating a program link for a micro:bit program, the link is not callable. Instead of the program the lab opens as if no program link was inserted. This is due to the misplacement of a mu sign (μ) as part of the robot identifier.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the Open Roberta Lab 
2. Select micro:bit as your robot system
3. Open the edit menu
4. Click on"create program link" (to create a program link for the default program)
5. Open a new tab or a new window in your browser.
6. Open the link that was copied to your clipboard.

**Expected behavior**
 - The program that was present in the Open Roberta Lab when the program link was created should be shown.

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Firefox 60 ESR

**Broken link example**
https://lab.open-roberta.org/#loadProgram&µbit&&NEPOprog&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22microbit%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22384%22%20y=%2250%22%3E%3Cblock%20type=%22robControls_start%22%20id=%22%7Cwyb/(51,q%25*G0FD1GKZ%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cmutation%20declare=%22false%22%3E%3C/mutation%3E%3Cfield%20name=%22DEBUG%22%3E%3C/field%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/program%3E%3Cconfig%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22microbit%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22138%22%20y=%2288%22%3E%3Cblock%20type=%22mbedBrick_microbit-Brick%22%20id=%221%22%20intask=%22true%22%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/config%3E%3C/export%3E 

**To solve the issue**
 - Instead of definining "μbit" as a robot identifier in the program link, the link works when replacing it with "microbit" and adding a second ampersand as delimiter before it.

**Fixed link example**
https://lab.open-roberta.org/#loadProgram&&microbit&&NEPOprog&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22microbit%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22384%22%20y=%2250%22%3E%3Cblock%20type=%22robControls_start%22%20id=%22%7Cwyb/(51,q%25*G0FD1GKZ%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cmutation%20declare=%22false%22%3E%3C/mutation%3E%3Cfield%20name=%22DEBUG%22%3E%3C/field%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/program%3E%3Cconfig%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22microbit%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22138%22%20y=%2288%22%3E%3Cblock%20type=%22mbedBrick_microbit-Brick%22%20id=%221%22%20intask=%22true%22%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/config%3E%3C/export%3E

bjost2s, 2018-12-21:
Could you please try it without explicitly copying the link? Once created it is already in the clipboard.

For Chrome I could not reproduce this problem although the displayed "μbit" looks strange.

philippmaurer, 2019-01-02:
"_Could you please try it without explicitly copying the link? Once created it is already in the clipboard._"
The link that is automatically saved in the clipboard is correct and therefore working fine, only the link that is shown in the textarea is broken.
This was reproducable on FireFox, Chrome and Edge. The program link did not work on IE, beause it was truncated after 2048 characters.

"_For Chrome I could not reproduce this problem although the displayed "μbit" looks strange._"
This did not work on all tested Browsers (FF, Chrome, IE, Edge) in Windows 10 and we should not expect it to. "μ" is not a legal character for URLS (RFC-3986-2.3) and should be URL-encoded.