OpenRoberta/robertalab #1103: Test collaboration of the Open Roberta Lab and raspian

bjost2s, 2018-12-22:
During the last months we have had a lot of problems concerning gcc compiler versions and linux distributions. We have to make sure that the lab is fully working on at least the raspberry pi, because this combination (lab and raspi) is used frequently in schools.

- [x]  2017-06-21-raspbian-jessy.img, tested via emulation including ArduinoSTL.h
- [x]  2018-11-13-raspbian-stretch.img, tested today on a raspi pi 2 including ArduinoSTL.h

bjost2s, 2018-12-22:
Conclusion: next Release 3.0.4-SNAPSHOT could also be deployed on raspis

pablopoo, 2018-12-27:
For this also could have, as an enhancement:
-An image ready to deploy
-An install script.
-An update script.
-A more advanced deploy option in a container service, like balena.io .

bjost2s, 2018-12-28:
We offer on our home page https://roberta-home.de a full working raspian image [here](https://www.roberta-home.de/lab/lokale-installation/) and I have just updated the detailed description in our github wiki [here](https://github.com/OpenRoberta/robertalab/wiki/Raspberry-Pi-2-3-and-the-Open-Roberta-Lab).    
The update script will follow asap.  
Also for container solutions you will find everything you need here in the repository (docker only).

closed 2019-06-26 by rbudde