OpenRoberta/robertalab #1105: How to connect a wedo robot?

collarad, 2018-12-29:
I would like to test my wedo robot with "open roberta lab" (https://lab.open-roberta.org/#). But, I do not understand how to connect the wedo robot. The button to "connect" is always disabled. 
I could not find any documentation in the wiki. 
Could you please provide me the required steps to connect my wedo robot to "open roberta lab" for testing? 
Thanks in advanced!!!

bjost2s, 2018-12-29:
Thank you for you question. As WeDo cannot hold and execute a program on its own, it needs to get commands from another device via Bluetooth. For this we have created so far two apps, one for IOS and one for Android which can do this sending and receiving commands via Bluetooth. Unfortunately they are not available in the app stores yet. 
But can find the Android app [here](https://github.com/OpenRoberta/app-android/releases). Once installed on your Android Smartphone and started,  you can click on the "connect" button and try to establish a connection to your WeDo, very similar than the Lego software is doing it.

collarad, 2019-01-09:
Thank you for the answer! Yes, via the mobile app I was able to connect to my Wedo robot.

closed 2019-01-09 by collarad