OpenRoberta/robertalab #1107: senseBox Integration

bjost2s, 2019-01-07:
The senseBox is a modular do-it-yourself environmental sensor station. Within the senseBox:edu you have sensors and actors available to create and program your own measuring station at home or in school. You can further publish your measurements in the [openSenseMap](https://sensebox.de/en/osem).
senseBox is based on the senseBox MCU with an ARM Cortex-M0+ processor SAM D21 family from Microchip.
- [ ] senseBox code generation, similar to Arduino
- [ ] senseBox  brick configuration, similar to Arduino
- [ ] senseBox  compiler workflow, similar to Arduino
- [ ] GUI blocks and toolbox
- [ ] GUI download executable to USB drive, similar to Calliope

For blocks and codegeneration please have a look at their [blockly](https://blockly.sensebox.de/ardublockly/?lang=en&board=sensebox-mcu) and in the [documentation](https://sensebox.github.io/books-v2/blockly/)

rbudde, 2019-01-28:
all tasks are finished, but there remain the following problems, about which a decision is needed:

- [ ] decide which blocks should go to beginner and which to expert toolbox. Move them.
- [ ] implement the SD card block after decision where to put the filename
- [ ] implement checker. Especially the send data block needs a test, that only sensor data is connected to that block. Now the server throws an exception. That is not user-friendly.
- [ ] write visitor tests

mariopesch, 2019-01-28:
Would like to help. Is there a dev preview online? I have tried to download the feature/sensebox branch and build it without any success. 

Error message when running `mvn install clean` on MacOS 10.14.2 and JDK 11
`class file for com.sun.beans.introspect.PropertyInfo not found`

VinArt, 2019-01-28:
@mariopesch it will be available tomorrow at test.open-roberta.org. Please try java 8, since 11 may have some classes removed and is in general not used by us.

VinArt, 2019-02-18:
- [ ] Buzzer did not work with tone() function but works fine with digital/analogWrite functions
- [ ] Off block for RGB LED produces empty line of code
- [ ] OpenSenseMap should be declared as a pointer for later instantiation or instantiated in place (2 phases for configuration variables generation needed).
- [ ] OpenSenseMap declaration stays even after the block was removed.
- [ ] Find out what is wrong with BMP280 (in regards to the data it delivers)
- [ ] Connection problems to OpenSenseMap persist (library problem? Connection quality problem?)

VinArt, 2019-02-18:
After receiving a reply about the buzzer a final decision will be made, currently the buzzer works, but has a fixed frequency.

VinArt, 2019-02-19:
I added the frequency to buzzer but it behaves strangely outside of certain range of frequencies. The tones are different, but not necessarily the correct ones. For my hearing 3kHz sounds higher than 8kHz and 50Hz lower than 30.

bjost2s, 2019-02-19:
- [ ] the download popup should be structured more nicely with same margins for all steps

bjost2s, 2019-02-20:
the current state is worth to be published as a beta version, so I add a preliminary relsoved label. For the next release we should have a look at the remaining issues (currently not ticked) in best case in a new issue.

VinArt, 2019-02-21:
- [ ] If the ID of the sensor is not put in the configuration block, wifi and sensor blocks configured - non compilable program is generated
- [ ] LED configuration blocks should probably not be fixed
- [ ] Reset timer should also have only timer 1 or get time should be able to get time from all 5 timers
- [ ] Parameter for g++ crosscompiler (for arduino builder) is wrong. Everything works, because the system compiler is used
- [ ] Other tool paths should also be checked
- [ ] Factory creation should accept the visitor, so to reduce the amount of factories needed. As well as the file extension.
- [ ]  WiFi bee MUST be on XBEE1 and SD on XBEE2
- [ ] Sensors measure values in actual units, so upper and lower border are defined by hardware -> put it in documentation.

VinArt, 2019-02-21:
Frequency of the buzzer is a bit odd:
```
#define  h     4064    // 246 Hz
#define  c     3830    // 261 Hz
#define  d     3400    // 294 Hz
#define  e     3038    // 329 Hz
#define  f     2864    // 349 Hz
#define  g     2550    // 392 Hz
#define  a     2272    // 440 Hz
#define  h     2028    // 493 Hz
#define  C     1912    // 523 Hz
#define  E     1518    // 659 Hz
#define  F     1432    // 698 Hz
#define  R     0       // Definiere eine Note als Ersatz für eine Pause
```
[from here](https://sensebox.github.io/books-v2/edu/de/projekte/Tonangriff.html)

bjost2s, 2019-02-21:
@VinArt as this task is already published, this issue is under review and as already written in the last comment we should consider: "For the next release we should have a look at the remaining issues (currently not ticked) in best case in a new issue."
Please create a new issue, so that we could close this with the already published state. Thank you

bjost2s, 2019-02-21:
"* LED configuration blocks should probably not be fixed" I think they should, because this two leds are fixed on the board with port 7 and 8 , so no need for the user tho define them on their own. See also WeDo inbuild configurations blocks like buzzer or key

VinArt, 2019-02-22:
I created #1138 with a list of things still needed to be done for the finilising sensebox, so I will close this one.

closed 2019-02-22 by VinArt