OpenRoberta/robertalab #1109: USBProgram: Make custom IP server / port persistent

pablopoo, 2019-01-07:
**Is your feature request related to a problem? Please describe.**
Users have to enter custom server every time the app is open

**Describe the solution you'd like**
Make this setting persistent.

boonto, 2019-01-11:
Thanks for the request.

For persistent changes in the USBProgram we have to consider how we integrate it without changing the user experience or confuse users.
E.g. one pupil could activate the option and close the USBProgram, the next pupil starting the program could not connect to the default lab.open-roberta.org because it does not know about the setting.

So we'll have to think about a nice way to realize this.

boonto, 2019-03-12:
The last 5 addresses are saved now, you just have to press on the advanced options dialog and the last used one should already be filled in.
Preview:
[image]

VinArt, 2019-03-26:
Is this merged to develop?

boonto, 2019-03-26:
Yes, should be: https://github.com/OpenRoberta/robertalab-usbprogram/commit/f3c154e700984573b1478240872d5494eca25fcb

closed 2019-03-27 by VinArt