OpenRoberta/robertalab #1122: Send message block (SendRemoteNumber) keeps resetting to channel number 0 [NXT]

GymNorfEhm, 2019-02-13:
**Describe the bug**
 - The Send Message block that allows sending data via bluetooth seems to be broken (at least when programming for NXT). The channel number keeps resetting to 0 on each compile/transfer aswell as each time you open the source view or update the source view.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to lab.open-roberta.org
2. Create a program for NXT
3. From NEPO-Blocks expert add a 'send message' block (found under Messages)
4. Change 'over channel' to anything but 0
5. Compile/transfer your code, or open source view, or refresh source view
6. See channel number resetting to 0

**Expected behavior**
 - Selected channel number is supposed to stay at selected channel number/

**Device information**
 - Type Desktop
 - OS: Windows 10
 - Browser: Firefox 60.5.1, but also tested on Chrome and Edge

GymNorfEhm, 2019-02-13:
On further investigation the 'send message' block seems to reset completely. The message type also resets from String to Number when opening or refreshing source view and on compile/transfer.

VinArt, 2019-02-22:
@GymNorfEhm thanks for testing, we indeed forgot to supply the information back to the client, so it was defaulting to 0 and Number. It should be fine and will be released in the next version.

VinArt, 2019-02-22:
So this fix for the NXT has cross-effect for EV3, namely the datatype (which is fixed to String) disappears from the block, though the generated code is correct.

VinArt, 2019-02-25:
EV3 datatype disappears no more, the only thing to fix are tests and this will be then done.

simonedegiacomi, 2019-03-09:
The bug appears also on the block to receive bluetooth messages: the message type and channel get resetted when opening the source code view.

closed 2019-06-12 by bjost2s