OpenRoberta/robertalab #1124: Bob3 start program button behaves not user-friendly

rbudde, 2019-02-14:
**Describe the bug**
the run program button, after pressed, becomes inactive for a short time and is enabled very soon, even before the message "program will start" appears. This will confuse users. Pressing the button a second time gives the error message "robot not active", which is confusing too.

**To Reproduce**
write a small program and press the run button (twice ...)

rbudde, 2019-02-15:
we have the same problem with ALL arduinos.

bjost2s, 2019-03-04:
The problem is at server side. Once the run button is pressed, the gui sets the run button correctly to disabled. The program starts to be processed on server side. Here we change the robots state to "busy", when the compiler workflow has succeeded. During the time a regular (<3s) ping is asking for the robots state and gets the correct info "wait" so that the gui switches back to enable the run button. 
We should introduce a new robot state on server side e.g. "compiling" which will return "busy" to the gui but accepts a new download like "wait".
On gui side we could only interrupt the ping until we got a response from the compiling (and download).

I would prefer the server solution, because with the new state we could block the robot for other run commands (only interesting if more clients are connected to one robot, used in schools for ev3 afaik)

closed 2019-03-26 by VinArt