OpenRoberta/robertalab #1125: set BOB3 board revision to V1.03 / problems with arm sensor

springob, 2019-02-15:
**Describe the bug**
 - latest SMD BOB3 have problems with arms, due to an really old revision switch...
  there are no V1.02 BOB3 out in the field (I think just the first BOB3 Beate build herself, but I am not sure)
  -> so we can set the revision **always to 1.03**

**FIX** in file /OpenRobertaParent/RobotArdu/resources/libraries/Bob3/BOB3.cpp:
void Bob3::init() {
  _bob3_revision = 103;  // V1.03
  leds_init();
  analog_init();
  leds_set_RGBx(1, OFF);
  leds_set_RGBx(2, OFF);
}

VinArt, 2019-02-20:
Does this not depend currently on:
```
  uint8_t sig2 = boot_signature_byte_get(4);
  if (sig2==0x0a) _bob3_revision = 102;  // V1.02
  if (sig2==0x0f) _bob3_revision = 103;  // V1.03
```
So.. If I understand this correctly, all the BOB3s out there will return 0x0f and the revision will be set to 103?

springob, 2019-02-21:
neither - the signature check is obsolete, latest SMD Variants are Version 1.10 and the signature check will result in value 102, which is very bad!
Resolution: Drop the signature check and always set version to 103:
`_bob3_revision = 103;  // Set always to 1.03`

side note: There are very more Bobs with version 1.10 in the world then the tree year old early Hardware Beta Version 1.02 so you can ignore the impact on them...

closed 2019-03-26 by VinArt