OpenRoberta/robertalab #1130: Ev3dev broken with 3.1.0 (still works with 3.0.3)

ensonic, 2019-02-19:
**Describe the bug**
Ev3dev based robots receive a http 500 (internal server error) when starting a prora from the lab.

**To Reproduce**
Steps to reproduce the behavior:
1.Connect an ev3dev based lego mindstorms robot
2. Create a program (does not matter, e.v. drive forward 20cm)
3. Click play
4. The robot gets http 500 and disconnects

bjost2s, 2019-02-19:
Is it connected via the cable or via wifi?
Same issue on local machines or https://test.open-roberta.org?

rbudde, 2019-02-19:
Artem already solved that issue by stored the Python sources in the file system. We have merge his changes and then hotfix this tomorrow

rbudde, 2019-02-19:
@bjost2s Better assign it to Artem

bjost2s, 2019-02-19:
@rbudde please prepare the hotfix today if possible so that I can publish it in the evening or tommorrow morning, otherwise it will be published next week. @VinArt , please assign yourself.

closed 2019-03-07 by bjost2s