OpenRoberta/robertalab #1135: Lego Boost Support

gnac, 2019-02-20:
**Is your feature request related to a problem? Please describe.**
 - Add support for Lego Boost robots so that newer developers could use Open Roberta to program their robots without being tied to the proprietary Lego Boost App. The Lego Boost App is only available on certain platforms and is only supported by specific devices making programming difficult. There exist open source python libraries as well as other closed source applications that provide interfaces to Lego Boost, but they require a higher level of programming skill and experience to use.

**Describe the solution you'd like**
 - It would be great if OpenRoberta could be extended to support Lego Boost. Perhaps by utilizing the pylgbst library developed by Andrey Pokhilko (https://github.com/undera/pylgbst). This library could be wrapped in a server interface or some other glue layer to provide interaction between OpenRoberta and the pylgbst libraries. Alternatively, Picaxe provides an S2BApp that has an http interface that can be used to communicate with the robot, but this app is currently not open source. S2BApp generates files that can be imported into Snap and Scratch, but only Scratch 2.0 is supported and the current web app doesn't work with these blocks.

**Describe alternatives you've considered**
 - An alternative would be to create custom Blockly blocks and package blockly as a standalone app. This app could export python code that uses the pylgbst library. Likewise Blockpy could be used as well to generate python code from blocks.

**Additional context**
 - In order to communicate with the Boost Robotic hub, a  Bluetooth Low Energy connection is required. Some devices have these integrated, but otherwise the user needs an addon dongle such as
https://www.silabs.com/products/wireless/bluetooth/bluetooth-low-energy-modules/bled112-bluetooth-smart-dongle
 - The pylgbst interface is documented on its github page: https://github.com/undera/pylgbst
- it appears necessary for the user to have or use a Bluetooth Low Energy dongle to interface with the robot hub. I have limited testing 
- Using the s2bapp and its generated snap/scratch blocks, I was able to determine its supported http interface

The s2bapp starts a server on http://localhost:17327. 
The general status of the hub can be requested using
`http://localhost:17327/poll`
This also tells the s2bapp that "Scratch is Connected"
All commands take the form of
`http://localhost:17327/<command>/<parameter_0>/<parameter_n>`
Some commands provide a callback like feature using a wait id
`http://localhost:17327/<command>/<waitId>/<parameter_0>/<parameter_n>`
which can be polled for with this request
`http://localhost:17327/poll/<waitId>`
This will return the status with a _busy line. if that line contains the command id, the command is still in process.
```
    // Check for absence of : "_busy ... <waitId>;" in 
    // http://localhost:17327/poll/<waitId>
    var busyTextStartPosition=poll.search("_busy");
    if (busyTextStartPosition&lt;0) {
        return true;
    }
    var busyText=poll.slice(busyTextStartPosition);
    var busyTextEndPosition=busyText.search("\n");
    busyText=busyText.slice(0,busyTextEndPosition)+" ";
    var waitIdStartPosition=busyText.search(" "+waitId);
    return (waitIdStartPosition&lt;0);

```

s2bapp http interface
```

/**
 * @fn motorOn/<motor>
 *
 * @brief Turn specified motor on
 *
 * @param motor is one of [ "motor A", "m*otor B", "motors AB", "motor C", "motor D" ]
 *
 * @example http://localhost:17327/motorOn/motor A
 */

/**
 * @fn motorOff/<motor>
 *
 * @brief Turn specified motor off
 *
 * @param motor is one of [ "motor A", "m*otor B", "motors AB", "motor C", "motor D" ]
 *
 * @example http://localhost:17327/motorOff/motor%20A/
 */

/**
 * @fn motorOnFor/<waitID>/<motor>/<timeInSeconds>
 *
 * @brief Turn specified motor on for a number of seconds
 *
[... 164 lines cut ...]
 * @return sensorA value of sensor A
 * @return sensorB value of sensor B
 * @return sensorC value of sensor C
 * @return sensorD value of sensor D
 * @return sensorE value of sensor E
 * @return sensorF value of sensor F
 * @return tiltx1 88
 * @return tilty1 0
 * @return current1 244
 * @return voltage1 2.9
 * @return button1 green hub button, true or false
 * @return distance 0-100
 * @return color color code of LED. 0-255
 * @return tilt 4
 * @return isTilted/any true or false
 * @return isTilted/up true or false
 * @return isTilted/down true or false
 * @return isTilted/left true or false
 * @return isTilted/right true or false
 * @return _busy <waitID> returns wait id of any running process.
 * @return _success message indicating s2bapp is succesfully connected to boost robot.
 * @return _problem message indicating problem connecting to robot.
 */

```

gnac, 2019-02-20:
Heres a list of all the lists used by the s2bapp interface:
```
          "motorAll": [ "motor A", "motor B", "motors AB", "motor C", "motor D", "all motors" ],
          "motorDirection": [ "this way", "that way", "other way" ],
          "motor":    [ "motor A", "motor B", "motors AB", "motor C", "motor D" ],
          "tiltDirAny": [ "any", "up", "down", "left", "right" ],
          "lightColor": [ "pink", "purple", "blue", "sky blue", "teal", "green", "yellow", "orange", "red", "white", "off", "random" ],
          "forwardOrBackward": [ "forwards", "backwards" ],
          "leftOrRight": [ "left", "right" ],
          "fBLR": [ "forwards", "backwards", "left", "right" ],
```

bjost2s, 2019-02-21:
Thank you very much for your detailed description. 
We have already published two apps, one for IOS and one for Android which are used to communicate between the browser (javascript) and WeDo, which maybe works similar than boost. Do you know if there are open source libraries for native Android and/or IOS available? That would make things much easier.