OpenRoberta/robertalab #1140: Block "in list get # variable" doesn't work

the-matzup, 2019-02-25:
**Describe the bug**
While using the block "in list get #" with a variable, it will create an error-message, if i want to run on EV3 leJOS 0.9.0 

**To Reproduce**
Steps to reproduce the behavior:
1. Create List
2. Create a number-variable
3. Use "in list get # variable" wherever you want
4. Connect to an EV3
5. Click on "run"
6. See the error-message

**Expected behavior**
Run the Code on EV3 without errors. 

**Screenshots**
[image]

**Device information**
 - Chrome, MacOS, 
- EV3 leJOS 0.9.0 (2018-04-11), OpenRoberta 2.6.2,

**Additional context**
- Works fine without errors in simulation view!
- Doesn't work to run on real EV3.

VinArt, 2019-02-25:
the actual issue is in the variable declaration: since all of them are declared as float, it is then impossible to use them as an index in any functions that expect int. Casting solves the problem, though certain behaviours may be occur not as expected when actually a float is given.

the-matzup, 2019-02-25:
Okay. Will there be a fix in the next release? 
Because, it is not possible to cast a number from float to int in open-roberta, is n't it? 

In addition, the pupils expect that they can use the numbers there. The same problem also exists with the counting variable i.

bjost2s, 2019-02-25:
@VinArt, as lists are used in all kind of robot systems, please check if this is a problem for other systems. too and document it here.  After this please make sure that we have corresponding unit and integration tests for this.

bjost2s, 2019-02-27:
@VinArt , please see comment above and make sure that this issue is part of the next releases project

the-matzup, 2019-02-27:
btw: No problems with "Calliope Mini", Nepo4Arduino, mBot, ...
Lists and variables are declared there as doubble

[image]

the-matzup, 2019-02-27:
But still problems with EV3, even if lists and varibals are now declared as float...

[image]

VinArt, 2019-02-27:
@the-matzup thanks for testing. Indeed, this is solely a java problem (to the best of my knowledge), since add/get/remove methods accept only int. Whereas for other languages this is not the case. For example, for C++ if one would try to set an element of the list at index 1.5 still the element at index 1 is set. I think this is due to casting... In C++ we are using the given double to advance the iterator. Iterator advance accepts Distance, which is typedef'ed as difference_type, which, in turn is typedef'ed as long. So there is a cast from double to long, which provides the needed integral part. this will also mean that 1.9(9) would result in operations on element 1. Python is not typed, but I think there is a similar cast somewhere under the hood. This I will test further though.

the-matzup, 2019-02-27:
@VinArt Thanks. Interesting. Didn't know the differences. 
Let me know if I can help with further tests.