OpenRoberta/robertalab #1141: Old picture of calliope mini at Roboter-Configuration

Thorlogy, 2019-02-26:
**Describe the bug**
 - I expect the latest picture of calliope mini at Robot-Configurtaion 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Open Roberta Lab.'
2. Click on 'RobotKonfiguration'
3. See old version of Calliope mini

**Expected behavior**
 - New version of Calliope mini.

**Screenshots**
[image]

**Device information**
- all devices

bjost2s, 2019-02-27:
Todo:
Merge our background picture with the new calliope version as background. The picture above is not suitable for this, we need a new svg with only calliope, no labeling

VinArt, 2019-04-01:
As I understand in 0b3af4d1ef95af0fc5e37ca502c92b27ea105ea5 new backgrounds were uploaded, and on lab server I see currently different ones in configuration, so I think this is resolved.

closed 2019-06-12 by bjost2s