OpenRoberta/robertalab #1142: No Simulation within IE

Thorlogy, 2019-02-26:
**Describe the bug**
 - With IE the OR Simulation is not working.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'open-roberta.org'
2. Click on 'SIM'
3. Nothing happens

**Expected behavior**
 - Open Roberta Simulator shows up and simulation is running

**Screenshots**
 - Nothing to show because nothing happens.

**Device information**
 - Type Desktop
 - OS: Windows 10 Enterprise 2016 LTSB, Version 1607
 - Browser Internet Explorer 11, Version 11.2485.433.0, Updateversion: 11.0.85

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2019-02-27:
Unfortunately IE 11 only supports ES5, not ES6
We have to update the following lines
`var blocklyprograms = programs.map(x => BUILDER.build(x.javaScriptProgram));` 
` programEvals = programs.map(x => new ProgramEval());` 
which are causing an error

bjost2s, 2019-02-27:
Are you using IE frequently?
Maybe you would like to become one of our testers?

closed 2019-03-27 by VinArt