OpenRoberta/robertalab #1143: Enable SSLCARevocationCheck for better security?

EvonX, 2019-02-27:
In the apache2 config file (https://github.com/OpenRoberta/robertalab/blob/ed5306ca36aae28d5679ace3666b9b5b8d22ff8a/Docker/TestSystemSetupTemplate/openroberta/conf/apache2/dev.open-roberta.org.443.conf) , SSLCARevocationCheck is set to none. However, the Apache official document recommends to set it to chain (https://httpd.apache.org/docs/2.4/mod/mod_ssl.html#SSLCARevocationCheck). I wonder if there is any security concern if SSLCARevocationCheck is set to none. Thanks.

VinArt, 2019-03-27:
There maybe an issue when setting this to true, since if the intermediate CA certificates are not imported, then the revocation check fails. For example, when signing commits with S/MIME that has intermediate CAs in the chain github cannot verify that the certificate was revoked. I fear that the same thing may happen to the SSL certificates for the web server.

rbudde, 2019-06-26:
revocation is handled by the "outer-most" Fraunhofer proxy and need not be handled by our apache.

closed 2019-06-26 by rbudde