OpenRoberta/robertalab #1157: Redesign of colour processing from frontend to code generation

bjost2s, 2019-03-10:
**Is your feature request related to a problem? Please describe.**
Colours are handled in the robot plugins now. This is not really needed and causes a lot of dead or unknown code.

**Describe the solution you'd like**
Different to #1071 I would propose a general solution in the robot project. This could be easy, because from blockly we get always the same information, the colour hex value, no matter which robot system. In the code generation we can then evaluate the colour and write the corresponding colour value as it is needed in the libraries, e.g. RGB, names or simple numbers.

**Describe alternatives you've considered**
#1071 is an attempt but seems to be not really suitable. Main problem is the conversion via enums which leads easily to bugs

bjost2s, 2019-03-21:
We should add the color mapping for bob3, ev3 and nxt again with the help of a yml file e.g.

VinArt, 2019-03-25:
in b5d69abb8708f70683bbc728ae643b5f10784dc9 the RGB macro was converted to an inline function. Should the macros that do the reverse (RCHANNEL, GCHANNEL, BCHANNEL) be also converted to inline functions?

bjost2s, 2019-03-25:
Yes, but we should have first a plan for this. There are a couple of still not working list blocks, related to this.