OpenRoberta/robertalab #1160: Wrong string escaping while compiling for EV3 (leJOS)

simonedegiacomi, 2019-03-13:
**Describe the bug**
String of text blocks aren't escaped correctly while generating the Java code for leJOS.

**To Reproduce**
1. Create a simple program to display a text with a slash in it
[image]

2. Try to send the program to the EV3

A dialog appears saying "Your program has errors so it cannot be compiled."

If you have a look at the generated code you can see that the slash character was incorrectly escaped:
[image]

**Expected behavior**
In this case there is no need to escape any character, the server should compile correctly the code and transfer the program to the EV3.

**Device information**
EV3 with leJOS 0.9.1 or 0.9.0

VinArt, 2019-03-14:
@simonedegiacomi Hello, thanks for checking this out, I wanted to ask a couple of things, before we merge this. Namely:

1.  org.apache.commons.lang3.StringEscapeUtils is deprecated, maybe use org.apache.commons.text.StringEscapeUtils instead if you are anyway changing the usage?

2. let's take a look at the type hierarchy of the language visitors:
[image]

your change changes the visiting of the String constant in AbstractLanguageVisitor class, which is at the very top of the hierarchy, so if noone overrides this method it will have the effect on all the sub-classes. Let's see if this is the case:

[image]

So visiting of the string constant is overriden by Calliope and Microbit, Calliope indeed has it's own way of representing them, namely feeding any string to the ManagedString class constructor and hoping that that will deal with everything. Microbit on the other hand is a different story, the override is simply a copy of the method from AbstractLanguageVisitor. Let's assume that the current implementation is wrong. So if you change it in AbstractLanguageVisitor, then it will stay incorrect for Microbit.

There are a couple of questions: is the escaping of the string the same for all languages? For example do we need to escape the same way for Java, C++ and Python? If yes, then the override from MicrobitPythonVisitor should be removed and the common implementation should be put in AbstractLanguageVisitor. If not, then we would need to distinguish this escaping by language and probably have correct (separate) implementations for each of them.

3. These kind of errors should be found by the integration tests, currently, I think, there is no manual on where to put the XML and what to do with it, so it would be taken by the integration tests, but soon there will be one.

So apart from the integration tests, it would be nice if you could check if the proper escaping is done in all cases and that there no unwanted side-effects.

Cheers.

simonedegiacomi, 2019-03-14:
Hi, thank you for taking the time to explain the sitation. I will review the changes as you suggested.
About the integration test, you mean that soon there will be a manual that specifies where to put the new test or that there will be the actual test?

VinArt, 2019-03-14:
@simonedegiacomi probably both, but I will update you as I figure it out.

simonedegiacomi, 2019-03-14:
ok, thank you!

I've checked the escaping sequences used in Python and, given that MicrobitPythonVisitor uses the `'` character to delimit strings, it is necessary to keep using escapeEcmaScript (which escapes both  `'` and `"`).
So I separated the String escaping from AbstractLanguageVisitor to the specific ones.

VinArt, 2019-03-15:
Hi, @simonedegiacomi 

I have taken a look at your branch, please consider the implementation in MicrobitPython visitor for removal, since if the behaviour for all python-based robots should be the same, then it should be implemented in AbstractPython visitor, I think. Also if all other robots should have the ecma-style escape, maybe it makes sense to just leave it in AbstractLanguage visitor, not to have copies in many places.

Another thing, the build fails due to tests failing.

Cheers.

simonedegiacomi, 2019-03-15:
It makes sense to remove the implementation in MicrobitPython, since now it is the same as AbstractPythonVisitor.

I noticed now (through the tests) that the implementation in Microbit used single quotes instead of double quotes. If I remove the implementation in Microbit and also from Python (to keep the ecma-style in AbstractLanguageVisitor) the tests for Microbit will need to be updated (single -> double quotes). Is it okay? also, is it right that after these edits the java-style will remain overriden in the JavaAbstract and CppAbstract?

Thank you

VinArt, 2019-03-18:
hi, I think that double and single quotes in Pyhton have the same meaning, but that could be easily tested. 

About the overrides: as long as there is no code duplication that could be pulled to the super class it's fine.

simonedegiacomi, 2019-03-18:
ok, thank you!

Now there are only 3 overrides: Cpp and Java because there isn't a common super class (and the implementation is different from the one used in AbstactLanguage) and Calliope, which has a different implementation.

VinArt, 2019-03-18:
Merged, though in future, in code clean-up we will go through such implementations and delegate them.

PR #1161 (2019-03-18): issue #1160 - Fix string escaping for EV3 leJOS
Fixes #1160
closed 2019-06-12 by bjost2s