OpenRoberta/robertalab #1166: Wait until: get analog/digital value sensor destroys program

VinArt, 2019-03-18:
**Describe the bug**
 - When using wait until block with get analog or digital value sensor in Arduino plugin the program gets destroyed

**To Reproduce**
Steps to reproduce the behavior:
1. Go to lab
2. Click on arduino (any type)
3. Use wait until block with get analog or digital value sensor
4. Open generated code
5. Program gets destroyed

**Expected behavior**
 - Program is generated

**Additional context**
This happens due to OUT_DIGITAL and OUT_ANALOG modes not being declared. There are currently 3 GetSampleType enums which are redundant, since all the information they contain is contained in .yml files. Those should be completely removed and proper modes declared in .yml files.

As for the sensor blocks themselves: calliope implementation uses one class from OpenRobertaRobot project and another one from RobotMbed for PinGetValueSensor and PinWriteValue (which is an action) respectively. This should be factored out to the OpenRobertaRobot project and re-used for Arduino that currently has it's own classes for both pieces of functionality.

Removing current implementation from Arduino has another benefit of using the SensorMetadataBean and not the transformation from the sensor class itself.

bjost2s, 2019-03-21:
related to #1080, #1017

bjost2s, 2019-03-25:
Last commits from me should belong to issue #1170! Sorry

closed 2019-06-12 by bjost2s