OpenRoberta/robertalab #1170: Calliope simulation: wait-block fails

rbudde, 2019-03-20:
A program waiting for a button press doesn't work in the simulation, but works on the real device.

**Device information**
 - win laptop
 - chrome and firefox. Independent of browsers.

bjost2s, 2019-03-21:
For me it is working, e.g. with the following program
[image]

chrome on linux, windows 10

bjost2s, 2019-03-21:
the difference is in the xml:
new
`<field name="SENSORPORT">button_a</field><field name="SLOT"/></block>`
old
`<field name="SENSORPORT">A</field>`

but
the new version should **not** work! but the old, very strange

rbudde, 2019-03-22:
Remark: this problem occurs, when an old Calliope program is loaded from the database. Its wait-block is not decomposable and the simulator freezes (the actual wait block from the tool box is ok, simulation works).

closed 2019-03-25 by VinArt