OpenRoberta/robertalab #1171: Cannot generate a program when using ev3dev middle motor

kzagar, 2019-03-21:
**Describe the bug**

Program using middle motor cannot be generated on ev3dev (error message: "The program could not be generated.")

**To Reproduce**

Steps to reproduce the behavior:

1. Add a "middle motor" to "ROBOT CONFIGURATION", say as "Motor A" 
2. On the NEPOprog, add a step to control the motor, say "set motor port A speed % to 30".
3. Click on "Open/close source code view"
4. Error "The program could not be generated" appears.

**Expected behavior**

Program is generated. Expected code:

```
#!/usr/bin/python

from __future__ import absolute_import
from roberta.ev3 import Hal
from roberta.BlocklyMethods import BlocklyMethods
from ev3dev import ev3 as ev3dev
import math

class BreakOutOfALoop(Exception): pass
class ContinueLoop(Exception): pass

_brickConfiguration = {
    'wheel-diameter': 5.6,
    'track-width': 18.0,
    'actors': {
        'A':Hal.makeMediumMotor(ev3dev.OUTPUT_A, 'on', 'foreward', 'right'),
    },
    'sensors': {
    },
}
hal = Hal(_brickConfiguration)

def run():
    hal.setRegulatedMotorSpeed('A', 30)

def main():
    try:
        run()
    except Exception as e:
        hal.drawText('Fehler im EV3', 0, 0)
        hal.drawText(e.__class__.__name__, 0, 1)
        hal.drawText(str(e), 0, 2)
        hal.drawText('Press any key', 0, 4)
        while not hal.isKeyPressed('any'): hal.waitFor(500)
        raise

if __name__ == "__main__":
    main()
```

**Screenshots**

N/A

**Device information**

Chrome

**Additional context**

N/A

simonedegiacomi, 2019-03-22:
@VinArt I think I have a solution to this issue.
If you want I can make a pull request to see if my solution is acceptable, otherwise it's not a big deal (it was my fault that I didn't comment on the issue before working on it)

VinArt, 2019-03-22:
@simonedegiacomi we actually decided to do it a bit differently, since the underlying hal for ev3dev was used not really in a good way, so I removed the side completely.

VinArt, 2019-03-22:
@ensonic hi, the 1997d12 removes the passing of side to make motor methods

simonedegiacomi, 2019-03-22:
> @simonedegiacomi we actually decided to do it a bit differently, since the underlying hal for ev3dev was used not really in a good way, so I removed the side completely.

@VinArt got it 👍

ensonic, 2019-03-31:
This is fixed in the new release 3.3.0 of the OpenRoberta lab that now is online.

closed 2019-06-12 by bjost2s