OpenRoberta/robertalab #1172: Calliope: name clashes (e.g. "sum")

rbudde, 2019-03-21:
Using "sum" as variable name gives a redeclaration error (with the function "sum" from MicroBit.h:41) and the program cannot be compiled

**Expected behavior**
- this name clash and all other clashes of this kind should be avoided.
- if not possible, at least the compiler message has to be shown to the user (maybe he/she can then avoid the problem)

**General remark**
 - a clear concept of separating user-defined names and ora-defined names from "standard library" names is needed

bjost2s, 2019-03-21:
same for "distance" (discovered) yesterday ;-)

philippmaurer, 2019-03-29:
Same for "fdd" when using the 4-digit display on calliope. (Mentioned in a support ticket)

boonto, 2019-03-29:
Yes, "fdd" and "ledBar" are clashing. I forgot to add the underscore to the variable declarations.

boonto, 2019-03-29:
I fixed it for "fdd" and "ledBar", the others are still a problem.

bjost2s, 2019-03-31:
We should prefix user defined names in general, no way to keep the reserved words list in blockly up to date.