OpenRoberta/robertalab #1174: Incorrect tooltip on "colour with rgb"-block for some translations

philippmaurer, 2019-03-27:
**Describe the bug**
On some languages the tooltip for the "color with rgb" block says that the number range is 0 - 100 and on some others the range is specified as 0 - 255.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a robot that supports the "color with rgb" block, like Calliope
2. Go into expert mode, open the colours category and hiver over the "color with rgb" block
3. If you saw a range of 0 - 255: Switch to german and hover over it again
4. If yousaw a range of 0 - 100: Switch to english and hover over it again

**Expected behavior**
 - Both translations should show the same range

**Additional context**
 - On calliope the correct range is 255 and on all robots the default values are outside of the 0 -100 range. Therefore i concluded that the 0 - 255 rnage is the correct one.

philippmaurer, 2019-03-27:
- [x] I corrected all translations in the translation sheet by setting the range to 0 - 255. 
- [ ] This just needs to be pushed to the repo and then be released.

bjost2s, 2019-03-29:
I will commit it, but I think we should have a deeper look in all systems using the rgb block, maybe there is still another system mapping the values to 0-100

philippmaurer, 2019-03-29:
The following systems use the "color with rgb" block and i already checked the checked ones.
- [x] calliope
- [ ] nepo4arduino
- [ ] mBot
- [x] sensebox
- [ ] NAO

closed 2019-06-12 by bjost2s