OpenRoberta/robertalab #1179: Debug and Assert Block

bjost2s, 2019-04-04:
**Is your feature request related to a problem? Please describe.**
For all common blocks we would like to generate integration test programs more or less automatically. For this it would make things much easier, if we would have a common debug block and a common assert block, so that we do not have to distinguish between the different plugins/systems.

**Describe the solution you'd like**
 The common debug block should generate some output, depending on the system. E.g. for Calliope it could be easily mapped to the "write serial", for EV3 there is afaik no similar block availabe, so a new implementation like "System.out.print(...)" could be necessary.
The common block "assert" should take two arguments very similar to 
the "logic_compare" block, but as a statement with the result going automatically into the new debug block!

ensonic, 2019-04-07:
For the ev3 I see two options:
* collect those messages and return them with the NepoExitCode (see #1147)
* stream logs using a websocket (needs a lot more changes)

bjost2s, 2019-04-11:
additional to my commit message: microbit tested