OpenRoberta/robertalab #1180: USB program cannot transfer programs anymore

bjost2s, 2019-04-07:
**Describe the bug**
When pressing the start button, the program seems to be created, but never is transferred to e.g. an Arduino
**To Reproduce**
- connect an Arduino to the pc (Ubuntu)
- open the latest version of the USB program
- connect to the lab
- type the token in the browser
- create a program
- click the run button

**Expected behavior**
 - program should be transferred to the arduino (flickering leds while transferring)

**Screenshots**
[image]

**Device information**
 - OS: Ubuntu 18.04

**Additional context**
 When starting the USB program from the console, logging shows, that the log file cannot be created, see screenshot. It seems that this problems stops execution in a way that the program is not transferred.

VinArt, 2019-04-08:
I believe maven filters resources and substitutes ${user.home} with actual user home, who built the binaries, in this case /home/pwerner.

```
./src/main/resources/logback.xml:        <file>${user.home}/OpenRobertaUSB/OpenRobertaUSB.log</file>
./target/classes/logback.xml:        <file>/home/here_be_my_username/OpenRobertaUSB/OpenRobertaUSB.log</file>
```

So in pom there should be an exception for this resource, since logback takes this variable from JVM, as I understand from it's documentation (run-time substitution?).

boonto, 2019-04-08:
Yes the logback file should not be filtered.
Does it work with this build?
[OpenRobertaUSBLinux.tar.gz](https://github.com/OpenRoberta/robertalab/files/3054949/OpenRobertaUSBLinux.tar.gz)